coredevices/PebbleOS #64: Could not find a code format file or style guide for contributions

BTPankow, 2025-06-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Proposed feature request

I couldn't find any style specification for how to format new code contributions. Is there any style guide we follow and could it make sense to add a `.clang-format` if there isn't an equivalent already? Just would help to make sure code style stays consistent with contributions.

gmarull, 2025-06-19:
`.clang-format` pretty much aligns with the styleguide, but it still isn't great and the codebase doesn't follow it everywhere, so an initial attempt to enforce it was reverted. The idea is to tune clang-format and reformat the whole codebase at some point, then enforce it.

gmarull, 2025-06-19:
> That makes sense, where is the styleguide? I might be able to take a stab at a clang-format file to match with minimal codebase disruption on reformat.

There's none, just existing files as a reference really. So, the idea is that any clang-format should apply minimal changes to existing files (some may be unavoidable, and some areas may require a clang-format off/on).

gmarull, 2026-10-06:
Done. C code is formatted with clang-format, and CI enforces it since 27922ece5 (#2085). Python uses ruff. Both are documented in `docs/development/contributing.md`.

closed 2026-10-06 by gmarull