coredevices/PebbleOS #71: Add support for RC10K

gmarull, 2025-06-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Proposed feature request

Add support for RC10K

HalfSweet, 2025-07-14:
> [@HalfSweet](https://github.com/HalfSweet) merged your PR, however, board init code lacks RC10K initialization, please check.

In fact, RC10k is automatically initialized after startup, so don't worry about it

[image]

gmarull, 2025-07-14:
> > [@HalfSweet](https://github.com/HalfSweet) merged your PR, however, board init code lacks RC10K initialization, please check.
> 
> In fact, RC10k is automatically initialized after startup, so don't worry about it
> 
> <img alt="Image" width="1738" height="928" src="https://private-user-images.githubusercontent.com/60973476/465840862-9199cdaa-3cae-4e78-99d8-cee2b4002b55.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI0NzYwMTgsIm5iZiI6MTc1MjQ3NTcxOCwicGF0aCI6Ii82MDk3MzQ3Ni80NjU4NDA4NjItOTE5OWNkYWEtM2NhZS00ZTc4LTk5ZDgtY2VlMmI0MDAyYjU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE0VDA2NDgzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJjMGNkNWFjZjZmMjNlOTkzMjI5OGQxYTllYzg1ZTViOWUwNGRkMjBmMzVkNzM2ODBlMWNjNjE3MGVhOTM1YjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GF3exXH0h_cuCTe-bc5ts4CHgPP6XBw_47qVF0SVMyw">

There’s still code in board init that tries to initialize LXT

closed 2025-07-15 by gmarull