coredevices/PebbleOS #109: gbitmap_sequence: Add support for APNG_DISPOSE_OP_PREVIOUS, or document lack of support

Ongion, 2025-07-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Proposed feature request

Currently, gbitmap_sequence does not properly support APNG_DISPOSE_OP_PREVIOUS, which should set the buffer to whatever was contained in the _previous_ frame. This is used to reduce filesize in animations, and is so commonly supported that no existing APNG encoders that I could find offer an option to _not_ use it.

When encountering APNG_DISPOSE_OP_PREVIOUS, gbitmap_sequence silently fails to do the correct thing and instead renders the same exact frame it was currently displaying. This can lead to situations where an animation displays incorrect data, which has personally left me questioning my sanity, and required me to modify an existing APNG encoder so that it would never consider using this dispose op.

I'd like to request that proper support for APNG_DISPOSE_OP_PREVIOUS be implemented, such that existing APNG encoders will work properly.

### Alternatives

I understand that proper implementation of APNG_DISPOSE_OP_PREVIOUS support could, in the worst-case, double the size of the buffer required for gbitmap_sequences, and that this may be prohibitively expensive memory-wise.
I would also find it perfectly acceptable to just document this lack of support so that others do not fall into the same pitfall that I did. Perhaps gbitmap_sequence could also log an error when encountering APNG_DISPOSE_OP_PREVIOUS, so developers would be informed when running `pebble logs`.

ericmigi, 2025-07-12:
Great Heiko question, or @ajwitte or @bradtjgmurray

HBehrens, 2025-07-16:
Thank you for calling this out, @Ongion 

I agree with your assessment that a proper implementation of `APNG_DISPOSE_OP_PREVIOUS` does require another buffer (even a prohibitively expensive re-decompression of all previous frames could be inaccurate if there are multiple chunks with partial update per frame). I cannot conclusively proof that memory considerations was driving this decision originally, but if I were to re-implement it today (for the old models), I'd make this decision for exactly that reason.

That being said, I second your opinion: Improved error reporting (e.g. a logged warning once per instance of `GBitmapSequence`) and updated documentation would help people who find themselves in a similar situation in the future.

Do you happen to have an incompatible APNG handy that you're willing to share as a test case?

closed 2026-10-06 by gmarull