coredevices/PebbleOS #181: Switch to LPTIM as RTOS timer on SF32LB52

gmarull, 2025-08-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Proposed feature request

If we expect reasonable power consumption, LPTIM should be used as a tick source in SF32LB52. Now systick is used.

cameled, 2025-08-31:
Some operation, like qspi flash sector may block irq nearly 32ms. LPTIM ticks should be handle it.

cameled, 2025-09-01:
> > Some operation, like qspi flash sector may block irq nearly 32ms. LPTIM ticks should be handle it.
> 
> Won’t this just add latency? LPTIM counter keeps going anyway.

Yeap, just comment in here.

cameled, 2025-09-01:
Another point, currently pebblos tick is based on 1024, means 1 tick nearly 0.9765 ms. Also find there have a `new_timer_service_loop` relay on that, which use 1024 as prescaler.
Obelix currently using RC10K, 1 tick will be 1ms. Need to update related code.