coredevices/PebbleOS #208: Move away from `waf`

gmarull, 2025-08-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

The build system is `waf` based. Filing as a bug because `waf` is a bug on its own.

### Expected Behavior

Use better options, like, e.g., CMake combined with Kconfig.

### Steps To Reproduce

`./waf ...`

### Version

db31924fac9846e379ce55eaa172bd58f1567cf1

### Host OS

N/A

### Watch

Pebble Time (Snowy), Pebble Time 2 (Obelix), Pebble 2 (Silk), Pebble 2 Duo (Asterix)

### Anything else?

N/A

matthewwardrop, 2025-08-25:
fwiw: I've enjoyed using `meson` in various contexts.

gmarull, 2026-10-06:
Done. The firmware now builds with CMake (#1967), unit tests run on CTest (#1980), and SDK packaging moved off waf (#1983). waf was removed from the repository root in 2ddc5efbc. The only waf left is `sdk/waf`, which ships to app developers.

closed 2026-10-06 by gmarull